coseyfannitutti/mysterium #2: Switch plate files?

pigpenguin, 2020-02-29:
Thanks for open sourcing the design files, I'm a big fan of the board.

Was curious if the files for the switch plate were in the repository some where as I can't seem to find them.

lackmorrison, 2020-03-04:
Also build guide says that you need screw-in stabilizers because there is no plate. Can plate mounted stabs be used if there is a plate?

coseyfannitutti, 2020-03-04:
@pigpenguin I have added a .dxf file for the plate to the repository.

@lackmorrison The plate included in the kit does not support plate mount stabilizers. It is possible to allow for plate mount stabilizers, but you would need to make your own plate, as it requires a different cutout than what is on the one I have uploaded.

My choice to use screw-in stabilizers even with a plate is simply due to them being significantly more popular with custom boards. They are also widely considered to be much better than plate mount, and I agree with that.

Thanks for pointing that out, though. I have removed the "because there is no plate" bit. That was accidentally carried over from the Discipline build guide because that kit did not come with a plate.

closed 2020-03-05 by coseyfannitutti