coseyfannitutti/mysterium #5: windows cannot recognize USBaspLoader device

marklai1998, 2020-05-08:


coseyfannitutti, 2020-05-08:
If you ran both the 'make flash' and 'make fuse' commands without any errors, and it is still not working, then something on the board is not properly soldered.

marklai1998, 2020-05-08:
Thx for the reply, I'm a newbie in electronics, thus came here for help.

What I can tell is:

1. I swap out some parts since that's not common in my place. First one is the UMT1H4R7MDD, I swap out for a taller one, still 4.7uF 50V but its 85°C, not 105. Another one are those resistors, I cannot find 1/6w and used 1/5w instead

2. we found the led is soldered in the opposite way, causing the light won't on, but not sure if it will cause the board not booting

3. I tried to use WinAVR at first but it won't make. After switching to another AVR toolchain, it makes, but also come with some error.

make
[image]

make flash
[image]

make fuse
[image]

I'll keep investigating it, and get back to you in a few days

marklai1998, 2020-05-08:
When I use WinAVR, it won't even make
wanna make sure the make file is okay
[image]

coseyfannitutti, 2020-05-08:
You will need to [download and unzip this file](http://www.madwizard.org/download/electronics/msys-1.0-vista64.zip).
Copy msys-1.0.dll to the \utils\bin directory of your WinAVR folder

That will take care of the child died before initialization error.

marklai1998, 2020-05-09:
[image]
still no luck with the dll loaded

coseyfannitutti, 2020-05-09:
You don't need to run the 'make' command. You only need 'make flash' and 'make fuse'.

marklai1998, 2020-05-09:
I directly use `make flash` then `make fuse` but still no luck(using both a new chip and old chip)
I verified the soldered with a voltmeter, the USB port is working without any bridging(date + - pin are also connected)

[image]

coseyfannitutti, 2020-05-09:
Are you certain that your crystal is 16MHz / 18pF?

Something between your USB port and your microcontroller is not right, and this appears to be a physical issue on the PCB. Without having the board physically in my hand, it will be difficult for me to troubleshoot a physical issue.

It appears that the bootloader has been flashed correctly. The bootloader works, and the PCB design works. If you had this PCB made yourself, the minimum order was likely 5. I would recommend trying to rebuild if you cannot find what is wrong on the PCB.

marklai1998, 2020-05-09:
Yes, I'd have 5 board on my hand, I'll try to rebuild one in these few days.
Thx a lot, I'll update you after that

marklai1998, 2020-05-29:
sorry for a long silence. I've been back and forth around those 5 board these days. 
I've been following your instruction and focus on the USB port. Greate news is that I've made 2 board working. Since 4 of my friends and I started this project. I still need to get the rest of the board working.

1 of the board act weirdly despite the USB port is soldered correctly. It starts out okay, can use for a few hours without a problem, but it stops working somehow when I reconnect the cable. And the data pins show they are connected, but somehow with a fluctuate and high resistance(sometimes it is 300ohm sometimes it is 5-600ohm)

Klagelied, 2021-06-01:
Hi,

I had the exact same problem with the Unknown USB Device (Device Descriptor Request Failed) Code 43. Only difference is that the problem affected my Discipline Keyboard, not my Mysterium.
Bootloader flashing and Firmware flashing worked great.

Next step was soldering the switches. After I hit the reset button, the error displayed. The board was tested before the switches were installed by me. 
I triple checked my Soldering, on the USB-C connector as well as on the other components that were directly connected to the MCU or the USB-Port.

My luck was that I got hands on a notebook that did not showed me the Unknown USB Device (Device Descriptor Request Failed) error. It seemed that some Keys were stuck in action, like they were permanently pressed - Maybe it is a security feature by Windows to disable the USB-Device with such behaviour. Resoldering the problematic switches did the trick. My Discipline now works on any Windows PC. 

Good luck on solving your issue!

closed 2024-09-18 by marklai1998