coseyfannitutti/mysterium #10: How to add capslock lights in qmk

kikko233, 2020-08-05:
i find gpio 32 not be used，i want to use this gpio with caps light. But i don't know how to set up in qmk，please help me ,think you very very much

nchieng, 2020-08-19:
Disclaimer: I have not yet flashed any keyboards with QMK as they are all still enroute. I have looked at the documentation a little to learn.

Without knowing what board/mcu/driver you have, there's only so much one can assume.

This [LED Control](https://docs.qmk.fm/#/custom_quantum_functions?id=led-control) documentation pretty mail nails it. You might just need to figure out what GPIO 32 corresponds to for your specific board.

You will need to ensure a few things have been done:

1. config.h
```c
/* Mechanical locking support. Use KC_LCAP, KC_LNUM or KC_LSCR instead in keymap */
#define LOCKING_SUPPORT_ENABLE
/* Locking resynchronize hack */
#define LOCKING_RESYNC_ENABLE
```

KolCrooks, 2020-12-07:
@kikko233 did you ever figure out how to add a caps light? I was thinking about adding one to my keyboard.