coseyfannitutti/mysterium #11: JLCPCB problem with Gerber files

BenediktWoermann, 2020-08-24:
Hi!
I have planed to order a pcb of mysterium keyboard at jlcpcb, but the included zip in this repo isn't recognized / theres no gerber preview. At the discipline, all of this works great! I tried to make my own gerbers with kicad with the required settings specified by jlcpcb but nothing changed. Is this a bug in the pcb file or in the system of jlc?
Any help is greatly appreciated!
Note: I've not uploaded the files before July, as stated on their website, but today

[image]

[image]

BenediktWoermann, 2020-08-24:
Also, if I try to open the zip gerber in this repo with the gerber viewer in kicad, I get tons of error caused by unexpected chars in any of those gerberfiles...

coseyfannitutti, 2020-08-24:
It's a bug with JLC's gerber viewer timing out before it generates the preview. This can sometimes happen with large boards or large drill files, which take a long time to process.

You'll need to enter the dimensions manually, 347.8 x 146.8mm, for both the main PCB and the bottom plate PCB.

You unfortunately won't be able to see a gerber preview on JLC's site with Mysterium. If you would like to see a preview before ordering you can upload the .zip file to https://tracespace.io/

coseyfannitutti, 2020-08-25:
What version of KiCad are you using? I am not experiencing any errors opening the .zip with gerbview at all.
[image]

BenediktWoermann, 2020-08-25:
After loading file by file into gerbview i recognized that the drill files weren't loaded correctly, but I think this was my fault, because I used "open Gerberfiles" for opening the .pdf. If I use "open zip archive file" everythinng works in gerbview.
I also recognized, that if I leave out the drill files, jlc shows a preview, so maybe your right and they are just to big to process in time. I'll try ordering the way you suggested! But I have one more question: In the preview without the drillfiles I can see that the color of the pcb (and more importantly the color of the three lines at the top left) depend on the surface finish I choose. Is this really the case? Which one would you suggest to get this nice gold color seen in the build guide? Options are HASL (with lead), LeadFree HASL-RoHS and ENIG-RoHS. Thank you very much!

BenediktWoermann, 2020-08-27:
Think I figured it out, HASL is silverish and ENIG is golden. Nevertheless took HASL, because its around 20$ cheaper.

closed 2020-08-27 by BenediktWoermann