coseyfannitutti/mysterium #13: Adding LEDs

stevenvo, 2020-09-09:
Is it possible to add LEDs or any type of lighting to the board?

Kimo47, 2020-09-09:
LED strip can be added by connecting GND and +5v on the underside of the ISP headers. Not sure about programmability. 

source: u/thearctican @ [https://www.reddit.com/r/MechanicalKeyboards/comments/f5kbc6/cftkb_mysterium_transparent_grey/](https://www.reddit.com/r/MechanicalKeyboards/comments/f5kbc6/cftkb_mysterium_transparent_grey/)

debido666, 2020-09-09:
It uses an ATMEGA32A-PU. (V2 model)
I imagine (hope) one just needs to find the right pin for addressing and add/enable RGB functions in firmware.
Guess I'll find out in October. When mine arrives.

d1554573r, 2020-09-20:
There is a similar topic in the discipline issues section. I'm trying to get this sorted but I get an unrecognized USB device error after flashing the firmware. I guess we can use a pin already taken for rows or columns? 

Anyway, QMK supports only certain types of LED strips: https://beta.docs.qmk.fm/using-qmk/hardware-features/lighting/feature_rgblight

LEDs have to be enabled in rules.mk and configured in config.h by uncommenting LED part:

` #define RGB_DI_PIN A7
 #ifdef RGB_DI_PIN
   #define RGBLED_NUM 16
   #define RGBLIGHT_HUE_STEP 8
   #define RGBLIGHT_SAT_STEP 8
   #define RGBLIGHT_VAL_STEP 8
   #define RGBLIGHT_LIMIT_VAL 255 /* The maximum brightness level */
   #define RGBLIGHT_SLEEP  /* If defined, the RGB lighting will be switched off when the host goes to sleep */
// /*== all animations enable ==*/
   #define RGBLIGHT_ANIMATIONS
// /*== or choose animations ==*/
//   #define RGBLIGHT_EFFECT_BREATHING
//   #define RGBLIGHT_EFFECT_RAINBOW_MOOD
//   #define RGBLIGHT_EFFECT_RAINBOW_SWIRL
//   #define RGBLIGHT_EFFECT_SNAKE
//   #define RGBLIGHT_EFFECT_KNIGHT
//   #define RGBLIGHT_EFFECT_CHRISTMAS
//   #define RGBLIGHT_EFFECT_STATIC_GRADIENT
//   #define RGBLIGHT_EFFECT_RGB_TEST
//   #define RGBLIGHT_EFFECT_ALTERNATING
 #endif`

I bet the problem I got is because A7 is already used as a column pin, I struggle to understand which pins are BX, CX and DX as this naming doesn't match ATMEGA32A pinout schematic.

stevenvo, 2020-09-21:
the guide mentions D71 pin used for ISO, I wonder if we can use it to control the rgb...

d1554573r, 2020-10-01:
According to the schematic, only C0 & C0 are not taken but if you assign any of them to the LED's you'll get an unrecognized device. Atm I got fixed blue LED's and I would love to replace them with an RBG led.

this-snow, 2020-12-27:
> According to the schematic, only C0 & C0 are not taken but if you assign any of them to the LED's you'll get an unrecognized device. Atm I got fixed blue LED's and I would love to replace them with an RBG led.

Using Pin 23 - C1 works fine. Controlling the RGB with defined Keys in the Keymap works fine.