coseyfannitutti/mysterium #15: Question about standoff heights in Mysterium case.

nchieng, 2020-09-29:


dhanakane, 2020-09-29:
Are you taking into account that the top and bottom layers are only for the screws to pass through? The standoffs do not pass through all the layers of the case.

nchieng, 2020-09-29:
No, I'm not.
If the screws pass through Top-1 and Bottom,
How are these pieces held in place?

dhanakane, 2020-09-29:
The screws are only 4mm. So you have 4mm screw --> 14mm standoff <-- 4mm screw. About 2mm of the thread engages the standoff on each side. The top and bottom sides of acrylic have smaller holes that only allow the screw body to pass through but a standoff cannot.

nchieng, 2020-09-29:
I think I get it. Thanks for the explanation!

So the top and bottom layers are essentially getting clamped by the screw caps and the standoffs.

closed 2020-09-29 by nchieng

dhanakane, 2020-09-29:
Yes! And the acrylic layers the screw threads pass through. The standoffs
align the case and provide a body for the screws to thread into as well.

coseyfannitutti, 2020-09-29:
That table is great. So, yes the standoff area length is ~15mm, and 14mm standoffs are used because of the slight variance in layer thickness of cast acrylic sheets.

It is the same for the feet, with ~9mm standoff area length, so 8mm standoffs are used.

nchieng, 2020-09-30:
Thanks! That now makes sense to me.

Here's the same table for the feet.
[image]

nchieng, 2020-09-30:
@coseyfannitutti 
Sorry to keep this alive. Can you point me in the right direction for the clear bumpons users at the bottom of the case please?

dhanakane, 2020-09-30:
They’re very similar to these “3M Bumpon Protective Products SJ5312 Blister of 56”

They’re widely used in domestic applications.