coseyfannitutti/mysterium #19: Hello there, hope you are well. I'm having an issue with my Mysterium- the 7, U, J, and M keys are not registering at all. I replaced diodes 22 and 24 and ensured that pin 4 on the MC socket is soldered correctly. Any ideas on how I can fix this?

I0PNR, 2020-10-07:


coseyfannitutti, 2020-10-08:
Make sure the microcontroller is seated properly and that none of the pins going into the socket are damaged. Otherwise, it sounds like you've got some damage on that column somewhere, either a scratch deep enough to cut the trace, or a damaged pad underneath your solder. A jumper wire should fix this issue for you.

These keys are all chained together on the upper pad of all of those switch positions. The circuit is highlighted in the picture below.

To see where this chain is broken and where to place your jumper wire, you will need to also test your F6 key, which is also on the same circuit.

If your F6 key is working, the chain is broken between F6 and your 7 key, and a jumper between the upper pads of those two keys. If your F6 key is also not working, then the issue is between F7 and the highlighted pin on the microcontroller. If that is the case, a jumper between the upper pad of F7 and that pin on the microcontroller should fix it.
[image]

coseyfannitutti, 2020-10-08:
@I0PNR pinging you here because I caught a couple typos, and said F7 a few times, when F6 should be the key to be looking at.

IOPNR, 2020-10-09:
@coseyfannitutti Hello again Bryan, just now had the chance to look at the board followed your direction to success. I ran the jumper from F6 to 7 and the whole column is up and running. Thank you for your prompt response, clear direction, and willingness to help - Much appreciated!