coseyfannitutti/mysterium #20: Gerber ZIP file not recognised by JLCPCB

nchieng, 2020-11-10:
**Reproduction Steps**
1. Create a new account on JLCPCB
2. Start a new quote
3. Select option to 'Add your Gerber file'
4. Upload the Mysterium.zip file from the Gerber folder

**Expected**
File is uploaded and processed successfully and a preview is generated. The PCB details are automatically filled in with dimensions and thickness. 

**Actual**
File uploads complete, but a preview is never generated and the form is not populated. 

I've tried re-zipping all the Gerber files myself and uploading that and am still encountering the same. 

Could this be an issue with JLCPCB's parser, or could something be done with the Gerber files?

nchieng, 2020-11-10:
I've just tried the file on PCBway and it managed to generate a preview correctly.

closed 2020-11-10 by nchieng

coseyfannitutti, 2020-11-10:
Yes, it is simply an issue with JLCPCB's gerber preview parser sometimes timing out. Their preview that also populates dimesions automatically, seems to be heavily reliant on the drill file, which in this case is extensive due to how many through hole footprints are on the PCB.

It is fine to order as is from JLCPCB if you manually enter the dimensions.
PCB and bottom plate dimensions: 347.8 x 146.8mm