coseyfannitutti/mysterium #24: Mysterium ISO problem

Schwelles, 2021-01-27:
Hello,

I recently build a mysterium and when i try to flash ISO layout on it the whole 3rd row goes one to the right. So the ISO enter is on the DEL, the DEL is on END, the END moves to PG DN etc.
I quickfixed it by just moving that row 1 to the left and it seems to work fine, so at least i can use it with ISO. 

I wrote a mail with no reply and also opened an Issue on the QMK github.

Hope you can help, thx.

TobbenTM, 2021-02-23:
I have the same problem, tried messing around with the keymap a bit, but didn't reach any good solutions. Would be really good if either @coseyfannitutti or someone else with better QMK knowledge could take a closer look!

HalfHeuslerAlloy, 2021-04-08:
Having a similar issue with a board from mechboards. 3rd row shifted and also the left slash key doesn't respond. I can re-solder to the ansi format but I'd prefer a working QMK iso config if possible.

HalfHeuslerAlloy, 2021-04-09:
I think I've fixed it, at least for my board layout. The key map was slightly off. I've submitted a pull request to the qmk github. In the mean time here is the changes I made to C:\Users\Matthew Vaughan\qmk_firmware\keyboards\coseyfannitutti\mysterium\mysterium.h

copy and paste the relevant section.

#define LAYOUT_tkl_iso( \
    K00,      K02, K03, K04, K05, K06, K07, K08, K09,      K0B, K0C, K0D, K0E, K0F, K0G, K0H, \
    K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1A, K1B, K1C, K1D,      K1F, K1G, K1H, \
    K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2A, K2B, K2C,           K2F, K2G, K2H, \
    K30, K31, K32, K33, K34, K35, K36, K37, K38, K39, K3A, K3B, K2D, K3D,                     \
    K40, K01, K41, K42, K43, K44, K45, K46, K47, K48, K49, K4A,      K4D,           K4G,      \
    K50, K51, K52,                K56,           K59, K5A,      K5C,      K5E, K5F, K5G, K5H  \
) \
{ \
    { K00, _x_, K02, K03, K04, K05, K06, K07, K08, K09, _x_, K0B, K0C, K0D, K0E, K0F, K0G, K0H }, \
    { K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1A, K1B, K1C, K1D, _x_, K1F, K1G, K1H }, \
    { K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2A, K2B, K2C, K2D, _x_, K2F, K2G, K2H }, \
    { K30, K31, K32, K33, K34, K35, K36, K37, K38, K39, K3A, K3B, _x_, K3D, _x_, _x_, _x_, _x_ }, \
    { K40, K41, K42, K43, K44, K45, K46, K47, K48, K49, K4A, K01, _x_, K4D, _x_, _x_, K4G, _x_ }, \
    { K50, K51, K52, _x_, _x_, _x_, K56, _x_, _x_, K59, K5A, _x_, K5C, _x_, K5E, K5F, K5G, K5H }  \
}

#define LAYOUT_tkl_iso_7u( \
    K00,      K02, K03, K04, K05, K06, K07, K08, K09,      K0B, K0C, K0D, K0E, K0F, K0G, K0H, \
    K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1A, K1B, K1C, K1D,      K1F, K1G, K1H, \
    K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2A, K2B, K2C,           K2F, K2G, K2H, \
    K30, K31, K32, K33, K34, K35, K36, K37, K38, K39, K3A, K3B, K3D, K2D,                     \
    K40, K01, K41, K42, K43, K44, K45, K46, K47, K48, K49, K4A,      K4D,           K4G,      \
    K50, K51, K52,                K56,                K5A,      K5C,      K5E, K5F, K5G, K5H  \
) \
{ \
    { K00, K01, K02, K03, K04, K05, K06, K07, K08, K09, _x_, K0B, K0C, K0D, K0E, K0F, K0G, K0H }, \
    { K10, K11, K12, K13, K14, K15, K16, K17, K18, K19, K1A, K1B, K1C, K1D, _x_, K1F, K1G, K1H }, \
    { K20, K21, K22, K23, K24, K25, K26, K27, K28, K29, K2A, K2B, K2C, K2D, _x_, K2F, K2G, K2H }, \
    { K30, K31, K32, K33, K34, K35, K36, K37, K38, K39, K3A, K3B, _x_, K3D, _x_, _x_, _x_, _x_ }, \
    { K40, K41, K42, K43, K44, K45, K46, K47, K48, K49, K4A, K01, _x_, K4D, _x_, _x_, K4G, _x_ }, \
    { K50, K51, K52, _x_, _x_, _x_, K56, _x_, _x_, _x_, K5A, _x_, K5C, _x_, K5E, K5F, K5G, K5H }  \
}