coseyfannitutti/mysterium #25: Compatibility with Lenovo ThinkPad dock

sonic757, 2021-02-04:
I built a mysterium with the high profile case for a friend and he is having issues with keys being "stuck" down (not physically) when the keyboard is plugged into a Lenovo ThinkPad dock. Keys would repeat forever if held down then released, and stop when hit repeatedly again. This issue does not present itself when connected directly to any computers including the laptop WHILE it is in the dock. Is there any way to fix this assuming the dock is not faulty? I looked into other issues and it might be a QMK firmware issue, but any insight would be very helpful.

dinosv, 2021-02-15:
It seems to me like a QMK firmware issue. I had a keyboard without flashing working ok for about two months, I just flashed it and had the same problem. Now I have it connected directly to the computer.

starwader, 2023-01-23:
I have a similar issue, did you resolve it somehow? 
Both keyboard and mouse freeze from time to time. 
Not sure if it is only the issue with dock as I tested it with my old x61s thinkpad and it acts the same.

dinosv, 2023-01-24:
> I have a similar issue, did you resolve it somehow? Both keyboard and mouse freeze from time to time. Not sure if it is only the issue with dock as I tested it with my old x61s thinkpad and it acts the same.

Not really. I still use it with direct connection to the PC.

starwader, 2023-01-25:
I think this may be a problem with the Linux driver. I will investigate that when I have some time

strwdr, 2026-08-25:
Hi again, I was able to fix the issue by adjusting polling interval to 10 ms. Seems like the defaults have changed since coseyfannitutti stopped maintaining the repo. You can see the fix on my forked qmk_firmware repo, under 'fix/mysterium_adjust_polling_interval' branch. 
Please let me know if that solves all the issues with the keyboard as well, and I'll open up a PR to the original repo. I hope this helps :)

dinosv, 2026-08-25:
Thank you!, I'll test it out this week.

strwdr, 2026-08-25:
also, "NO_SUSPEND_POWER_DOWN = yes" in rules.mk seems to be helping with some freezes

dinosv, 2026-09-11:
It worked for the repeated keys! but now I have some freezes (with lost keystrokes), even with the  "NO_SUSPEND_POWER_DOWN = yes" fix.

strwdr, 2026-09-13:
I also experienced these additional freezes. What ended up fixing all issues was disabling the startup check along with increasing the polling interval to 10ms.

I opened a PR with this change, but QMK maintainers considered it too invasive, because disabling the startup check breaks OS suspend wakeup and the lower polling interval generally works fine on other ATmega32A boards. 

Because of that, this fix is best kept in user-space. 
This is the full workaround to add to `keyboard.json`:

```diff
   "usb": { 
       "vid": "0x6B62", 
       "pid": "0x8769", 
-      "device_version": "0.0.1" 
+      "device_version": "0.0.1",
+      "polling_interval": 10,
+      "no_startup_check": true
   },