coseyfannitutti/mysterium #26: Right side column (SW15,SW32,SW49,SW94) not working

lednax, 2021-02-13:
Hey @coseyfannitutti, I recently built a Mysterium (No iso) but the right side column isn't working. I tested for continuity based on the schematics and it appears there IS continuity but wanted to get your expert opinion. The rest of the kb works fine. 

Screenshot of the column: 
[image]

lednax, 2021-02-13:
I tested continuity again and I'm leaning torwards a software issue. Is the `default` qmk not good for v1?

this-snow, 2021-02-13:
Did you tested continuity up to the Controller? And continuity between the diodes, that are for this column. 
There is no software issue.

lednax, 2021-02-14:
Checked continuity again up to the controller and it was fine but then noticed a bent leg on the controller so I got it out an fixed that. Works fine now.

closed 2021-02-14 by lednax