coseyfannitutti/mysterium #37: Hi, please add the build file, it's missing on your website.

SpicyWiener69, 2024-09-21:
Thank you.

alexandre97costa, 2025-02-08:
Hi, this repo seems abandoned. If you still need a build guide, Pete from 42keebs has got you covered https://github.com/piit79/mysterium/blob/master/doc/MYSTERIUM_V2_BUILD_GUIDE.pdf

papierkorp, 2025-03-27:
Here are the files from his site per [archive.org](https://web.archive.org/web/20210602133830/https://www.cftkb.com/shop/mysterium-tkl-diy-keyboard-kit)

[MYSTERIUM+FLASHING+GUIDE.pdf](https://github.com/user-attachments/files/19486079/MYSTERIUM%2BFLASHING%2BGUIDE.pdf)
[MYSTERIUM+V2+BUILD+GUIDE.pdf](https://github.com/user-attachments/files/19486078/MYSTERIUM%2BV2%2BBUILD%2BGUIDE.pdf)
[MYSTERIUM+FLASHING+GUIDE+-+COMMAND+LINE.pdf](https://github.com/user-attachments/files/19486080/MYSTERIUM%2BFLASHING%2BGUIDE%2B-%2BCOMMAND%2BLINE.pdf)